nakujaproject/internship2022 #30: Research on camera.

Mike-Kimani, 2022-01-20:
Find the most suitable on rocket camera to record a video of the flight.

bytecod3, 2022-02-08:
An RF FPV camera with a TFT-LCD monitor gave good results during testing. It can be attached to the body of the rocket during flight to record and transmit flight video. The camera was able to transmit good stream up to distances of 1.1 km. 
Images of the test can be seen here - https://github.com/nakujaproject/N2Avionics-playground/commit/befeffd551427fd049c06108e2e3de7c9344d042

We noticed signal quality is great when antennas are upright than when they are pointed at each other

junn-hope, 2022-02-09:
The ESP32 CAM proved to have good resolution over a very short distance of up to 5m, however, this result is tied to the strength of the Wi-Fi to which it is connected to.

Beyond 15m, the camera lost its clarity. Abstracting from this results, I would not recommend it for use for N2- 500m apogee goal.

0x6f736f646f, 2022-03-29:
Done

closed 2022-03-29 by 0x6f736f646f