nakujaproject/internship2022 #43: Test send and receive over lora

0x6f736f646f, 2022-01-28:


bytecod3, 2022-02-03:
Lora was able to send and receive. Code uploaded on avionicsv2/LORA_TESTS/

bytecod3, 2022-02-07:
Done by this  https://github.com/nakujaproject/N2Avionics-playground/commit/1af6415088bc62bf34250a2641d6748b4dcc1741
Transmission was done using encoded JSON data. The problem is the speed of the packets. Some packets are being lost due to latency and signal spreading

bytecod3, 2022-02-08:
Fritzing diagrams for sending and receiving using Lora RF96 Module done here https://github.com/nakujaproject/N2Avionics-playground/commit/eb0c37cc47ffd391e9283701375c87f888a46157
All deliverables for this issue have been implemented

0x6f736f646f, 2022-02-09:
> Done by this [nakujaproject/N2Avionics-playground@1af6415](https://github.com/nakujaproject/N2Avionics-playground/commit/1af6415088bc62bf34250a2641d6748b4dcc1741) Transmission was done using encoded JSON data. The problem is the speed of the packets. Some packets are being lost due to latency and signal spreading

Don't send JSON. Send bytes or bits. This is more efficient.

0x6f736f646f, 2022-03-29:
We will use theoretical