nakujaproject/internship2022 #49: Inquire about JKUAT satellite

0x6f736f646f, 2022-01-28:


0x6f736f646f, 2022-01-29:
@junn-hope  provide the specs for this

junn-hope, 2022-02-01:
@0x6f736f646f  We met the person in charge of this satellite on Monday,31st January. He confirmed that our idea of routing our traffic through this satellite can be done by creating a Wi-Fi network around our test area delivered by antennas set up point to point from transmitting antennas at Tech House. The Wi-Fi shall set up by a router connected to the receiver antenna. We can then configure our ESP32s, RaspberryPi and PCs to the same network.

A setup cost of approximately Ksh 30,000 was identified to coater for the cost of setup and for bandwidth.

0x6f736f646f, 2022-03-29:
Will not be implemented

closed 2022-03-29 by 0x6f736f646f