nakujaproject/internship2022 #50: Design architecture for ground station

0x6f736f646f, 2022-01-28:


junn-hope, 2022-02-01:
@0x6f736f646f , @Mike-Kimani 
Open for review, this was our idea on how the architecture shall look like.

[image]

0x6f736f646f, 2022-02-01:
It is not yet done until approval

0x6f736f646f, 2022-02-01:
Can you outline the steps in each process e.g

In Manager
- Receive data from esp32 receiver
- Decode the data
- Store the data
- Display the data

0x6f736f646f, 2022-02-01:
> @0x6f736f646f , @Mike-Kimani Open for review, this was our idea on how the architecture shall look like.
> 
> [image]

Good job @junn-hope and @bytecod3

closed 2022-03-23 by 0x6f736f646f