nakujaproject/internship2022 #51: Inquire about Gegis drones

0x6f736f646f, 2022-01-28:


0x6f736f646f, 2022-01-29:
@junn-hope  provide the specs for this

junn-hope, 2022-02-01:
@0x6f736f646f 
I have spoken to the person in charge of the GEGIS drones, Mr Lazarus, he says the drones are available for  use. However, we have to request for them from the GEGIS department chair through our supervisor-Dr Aoki. I shall table this during tomorrow's meeting

An issue over KCAA regulation for our intended flying height was raised; as JKUAT is currently licensed to fly below 117m. I would suggest for test purposes, we keep within the regulated height.

shohei, 2022-02-01:
@junn-hope Thank you for the update. Can you draft the letter to request the drone use of GEGIS department?
For the KCAA regulation, I'm still in touch with the KSA for how we proceed with the flight waiver. We'll comply with the 400 feet regulation until we get the clearance from the authority.

junn-hope, 2022-02-01:
@shohei  Yes sir I shall draft the letter and deliver by close of business.

closed 2022-03-29 by 0x6f736f646f